AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #988: Battery Percentage doesn't show in power flow

seedoilz, 2023-07-03:
I love the new feature of power flow because this is very cool.
However, I have some problems:

1. I can't read the battery percentage of my MacBook as I use AlDente to replace the native battery icon. Maybe the percentage could be added to the AlDente's popover in power flow.
2. Sometimes Calibration may be interrupted by some things. And it won't show the recover button (Sometimes it does show). I hope may there is a way for me to choose the phase of Calibration.

Thanks. I love AlDente!

MatthiasKerbl, 2023-07-19:
Hi @seedoilz,

Thank you for your kind words. :)

1. You can add the battery percentage next to the battery icon in AlDente Settings/Appearance/Show Percentage.
2. In which specific situations does the recover button for the Calibration Mode not work?

Best,
Matthias

seedoilz, 2023-07-19:
Hi @MatthiasKerbl 
About the first point, I don't think putting the battery percentage next to the battery is very good looking.😂 
And for the second point, recently I find that only when I shut down my mac, the recovery of calibration will not work. I think it is not a big deal. So you could ignore it.