AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #990: Why resetting the SMC and NVRAM don't work after uninstalling Al Dente?

metebyte, 2023-07-04:
### Discussed in https://github.com/davidwernhart/AlDente-Charge-Limiter/discussions/983

<div type='discussions-op-text'>

<sup>Originally posted by **metebyte** June 27, 2023</sup>
Hi there Al Dente community,

As I mentioned on the title above, Al Dente still affects the charge process even after completely uninstalling it. Resetting the SMC and NVRAM work but only per boot. I mean I have to reset the SMC at every time I boot. Otherwise it doesn't work. Has anyone experienced something like this issue?</div>

metebyte, 2023-07-15:
Before closing this issue, I would like to provide some additional information to inform other users.

First and foremost, Al Dente is not a revolutionary application with extraordinary features. However, its promised functions are quite satisfactory in terms of efficiency. Nevertheless, you should consider some risks for the health of your device.

During my 3-month usage of Al Dente Pro, I encountered at least 20 instances of abnormal charging behavior, leading to frustrating situations of running out of battery. Despite following all visible guides and instructions, I couldn't find a solution, and eventually, I had to remove Al Dente. However, little did I know that after uninstalling Al Dente, the nightmare continued due to my charging functionality becoming completely disastrous. At last, suspecting damage to my battery calibration, I decided to replace the battery. Yet, neither the new battery nor reinstalling macOS on a fresh SSD resolved my laptop's charging issues. Despite trying all suggestions and solutions from relevant communities and information sources, I couldn't get any results. Feeling defeated, I began to suspect that the problem might lie with me, so I decided to try it on another device. As a result, when I performed the installation on a device where Al Dente had never been installed before, and after a short period of use, I encountered the same disappointing results. This made me give up worrying about this issue. I doubt that the development team is concerned about this matter in any way either.

As a humble piece of advice, if you love your computer as a developer, I strongly believe that you should keep Al Dente far away from your system.

With warm regards.

closed 2023-07-18 by metebyte

senhalil, 2023-07-18:
@metebyte any update on this issue? I was planning to try AlDente but after seeing your issue, I am having second thoughts.

MatthiasKerbl, 2023-07-19:
Hi @senhalil,

You can follow our discussion about this topic here #983.

Best,
Matthias

metebyte, 2023-07-19:
Hello @senhalil,

I hope this issue hasn't caused you to be biased against Al Dente. First and foremost, I must state that Al Dente is an application that has delivered efficient results based on its promised functions. In my opinion, the fact that people like me, who have encountered such issues, generally face similar problems across different platforms and devices does not imply that the application is harmful in any way.

However, if you'd like to hear a few opinions, there are a couple of points I'd like to mention:

1. You should not forget that Al Dente explicitly promises to provide protection rather than extending battery life.
2. The Pro version is a significant factor in terms of the application's working principle; by using the Pro version, you can avoid many headaches.

With best regards