AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #992: Top-up feature resets when charging is restarted

abdullahkhilji, 2023-07-09:
**Describe the bug**
Top-up feature resets when charging is restarted (not sure if its a bug or a feature request)

**To Reproduce**
Steps to reproduce the behavior:
1. Top-up battery level to 100%
2. Stop charging
3. Restart charging (top-up is now resetted and AlDente is now in sailing mode)

**Expected behavior**
Top-up should only stop when it is achieved

**Screenshots**
NA

**MacBook (please complete the following information):**
 - 16" Pro
 - Ventura
 - AlDente Pro
 - 1.22
 - List of all enabled/disabled settings in AlDente:

**Additional context**
NA

DamonHD, 2023-07-09:
I am happy with this behavour as-is for now.  When charging off-grid and my off-grid system starts to fade I am happy that AlDente stops the top-up.  I have been relying on this today for example.

As an alternative I'd be happy if AlDente could throttle back the top-up until not causing the source power to drop out, but that may be charge-control finesse beyond its reach.

Rgds

Damon

MatthiasKerbl, 2023-07-19:
Hi @abdullahkhilji,

This is the intended behavior of Tup Up. It is only supposed to start charging to 100% once until it is interrupted. You can read more about the Top Up feature on our blog "[Feature Explanation: Top Up](https://apphousekitchen.com/feature-explanation-top-up/)"

Best,
Matthias

closed 2023-07-19 by MatthiasKerbl