AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #993: Sailing Mode for Heat Protection

Elijas, 2023-07-11:
**Is your feature request related to a problem? Please describe.**
- Sailing mode is supposed to prevent repeated charging / stopping charging.
- Heat Protection does exactly that 33 degrees (stop) -> 32.9 degrees (start) -> 33 degress (stop)

**Describe the solution you'd like**
Degrees sailing mode (e.g. selectable 1-5 degrees) 33 degrees (stop) -> 31 degrees (start) -> 33 degrees (stop)

**Describe alternatives you've considered**
Using lower powered charging brick to limit power and therefore heat

**Additional context**
n/a

mojoba, 2023-07-17:
I second this feature request. 

I currently see rapid charge/discharge bouncing while connected to a 140w power supply and battery at or over 50% due to heat protection at 33c and battery at 32.9c.

Either a wattage limit while charging or a sailing mode for Heat Protection is ideal.

MatthiasKerbl, 2023-07-19:
Hi @Elijas and @mojoba,

You are correct, an improvement of the Heat Protection feature in this direction would make a lot of sense and we will look into it.
I will close this issue since we already have a feature request open for this issue here #675.

Best,
Matthias

closed 2023-07-19 by MatthiasKerbl