AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #998: Screen stays on while lid is closed and "Disable sleep until charge limit" is active

LackedMule1217, 2023-07-18:
**Describe the bug**
The screen stays on even when lid is closed and "Disable sleep until charge limit" is active.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings
2. Click on 'Disable sleep until charge limit' and enable it
3. Close the MacBook lid
4. Screen still stays lit. This drains battery.

**Expected behavior**
When I close the lid, the screen should turn off to conserve battery. And after it has charged to the target percentage, it should auto-sleep.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.
[image]

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608): A2779
 - macOS version (https://support.apple.com/en-us/HT201260): 13.4.1 (c)
 - AlDente Pro or AlDente Free?: AlDente Pro
 - AlDente Version?: 1.22
 - List of all enabled/disabled settings in AlDente:
[image]
[image]
[image]
[image]

olasun, 2023-07-20:
Can confirm this, reproduced on a machine described below.

- MacBook model (https://support.apple.com/en-us/HT201300): A2442

- macOS version (https://support.apple.com/en-us/HT201260): 13.4.1 (c)

- AlDente Pro or AlDente Free?: AlDente Pro

- AlDente Version?: 1.22

No need for duplicating identical screenshots as they already are in the original post.

dreit-p, 2023-08-19:
I'm really looking forward to the fix. I always find my laptop hot and not charged because of this bug.

dreit-p, 2023-08-27:
I have a workaround for those who are facing the same problem. Use the [amphetamine app](https://apps.apple.com/us/app/amphetamine/id937984704?mt=12) to block sleep while charging. In AlDente itself, disable this feature.

MatthiasKerbl, 2024-06-18:
Hi @LackedMule1217,

Good news, we have just released AlDente 1.27 which will now turn off the backlight of the display when the lid is closed and Disable Sleep Until Charg Limit is active.

Best,
Matthias

closed 2024-06-18 by MatthiasKerbl