AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1003: AlDente crashing on macOS 12.6

alessionossa, 2023-07-22:
**Describe the bug**
I get a crash after a couple of seconds I try to open AlDente on my MacBoon with macOS 12.6. It works fine on my other MacBook with macOS 13.4; however, I cannot update anymore the first one

**To Reproduce**
Open AlDente - it will crash.

**Expected behavior**
AlDente should not crash on launch 

**Screenshots**
/

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro (15-inch, 2016)
 - macOS version: 12.6.7 (21G651)
 - AlDente Pro
 - AlDente Version: 1.22 (40)
 - List of all enabled/disabled settings in AlDente:

**Additional context**
I can provide crash report privately if needed.

chrisgot, 2023-08-18:
I have the same issue on my 15" MBP Late 2016. I already reported it via E-Mail to the support.

jeanpascal395, 2023-08-20:
I join a similar problem with my 15'' MBP Late 2016. After I enabled the stats in the settings, the app crashes if I click on the icon in die menu bar. Function and display of the charging Icon in the menu bar are not affected. Maybe the crash behavior depends on the stat function?

MatthiasKerbl, 2023-08-21:
Hi everyone,

Thank you for letting us know about this bug. We are working on a fix for this issue and will release it with one of the next updates.

Best,
Matthias

MatthiasKerbl, 2023-10-10:
Hi everyone,

We have just released AlDente 1.22.3 which includes a fix for this issue. To get AlDente 1.22.3 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Cheers,
Matthias

closed 2023-11-09 by MatthiasKerbl