AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1017: Unable to re-activate license

udayk45, 2023-08-02:
**Describe the bug**
After upgrading to macOS Ventura, ai dente showed popup to recheck license. But it failed always. So tried re-installing but unable to activate the license after reinstall as well.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.
[image]

**MacBook (please complete the following information):**
 - MacBook model - Apple M1 Pro
 - macOS version - 13.4.1 
 - AlDente Pro or AlDente Free - AlDente Pro
 - AlDente Version - AlDente Pro

MatthiasKerbl, 2023-08-03:
Hi @udayk45,

Most of the time, this issue occurs because your device is unable to connect to our license server. Are you using a VPN or Firewall application that might block this connection? If so, please try to disable it and see if the activation works afterward. Furthermore, some countries are blocking the connection to the licensing server as well. Therefore, also try to use a VPN to connect to the licensing server through different countries.

Best,
Matthias

udayk45, 2023-08-03:
With VPN is started working again.

closed 2023-08-03 by udayk45