AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1021: [Heat Protection] make Power to 0, CPU downclocked, auto shutdown

Abirdcfly, 2023-08-15:
**Describe the bug**

When Heat Protection is turned on, it only compares the battery temperature without considering the battery power, which on my computer can easily result in 0 power, and when I realize that the power is almost 0 and will automatically shut down, often the cpu will be downclocked again because of the low power, a vicious cycle...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608) 13-inch, 2018, Four Thunderbolt 3 Ports
 - macOS version (https://support.apple.com/en-us/HT201260) 13.5 (22G74) 
 - AlDente Pro or AlDente Free? Pro
 - AlDente Version? 1.22.2
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-08-18:
Hi @Abirdcfly,

I am not sure I am understanding your issue correctly.
Could you explain it in more detail?
What do you mean by 0 power?

Best,
Matthias

Abirdcfly, 2023-08-18:
Apologize for my bad English, '0 power' referring to the percentage of battery power remaining as 0, low battery

MatthiasKerbl, 2023-08-18:
Hi @Abirdcfly,

No worries. First of all, try not to get down to 0% battery too often since it's not healthy for the battery. Also, if the Heat Protection feature gets triggered too easily, try to adjust the temperature in the settings.
Usually, the MacBook should not shut down when it is plugged in even if Heat Protection has paused charging. Does your MacBook shut down when it is plugged in and Heat Protection has paused charging?

Best,
Matthias

Abirdcfly, 2023-08-19:
> try to adjust the temperature in the settings.

Yes. That's what I'm doing right now.

> Does your MacBook shut down when it is plugged in and Heat Protection has paused charging?

Yes. Battery power slowly decreases until 0. Then it shuts off. I'm guessing this may have something to do with the fact that I'm plugged into a typeC monitor, which will provide 65W charging, which may not be quite enough.

MatthiasKerbl, 2023-08-21:
Hi @Abirdcfly,

> Yes. Battery power slowly decreases until 0. Then it shuts off. I'm guessing this may have something to do with the fact that I'm plugged into a typeC monitor, which will provide 65W charging, which may not be quite enough.

This could be the issue. If you have a power-hungry MacBook which uses more than 65 W over a longer period, the battery will get discharged. I am thinking of two options you have to counteract this issue. Firstly, you can use a more powerful power supply preferably the original one. Secondly, you can try to reduce the power consumption of your MacBook. Try to use our new beta feature Power Flow to keep track of the power consumption of the MacBook.

Best,
Matthias

closed 2023-11-15 by MatthiasKerbl