AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1022: Percentage on popup window

Glydric, 2023-08-18:
**Is your feature request related to a problem? Please describe.**
On standard macos battery icon you can see the percentage when clicked, this gives a nice look and a battery precise infomation only when wanted. 

**Describe the solution you'd like**
I would like same feature/option on the charging bar of aldente popup window

**Describe alternatives you've considered**
Maybe the percentage can be on the space between limit and discharge buttons

**Additional context**

[image]

dreit-p, 2023-08-19:
Also met this problem, I wanted to use the simple battery icon w/o percentage, but realized I can't get know the percent of the battery charge

mccarverone, 2023-09-09:
This would be greatly appreciated. Thanks

Bat-Maniac, 2023-09-29:
It would be great if the current battery level and percentage could be displayed in the popover instead of in the menu bar. Just like the native battery indicator on macOS.

If show percentage in menubar is deactivated in settings, there is no way to display the hardware percentage. So it would be nice to add an option to show it in popover instead of menubar.

MatthiasKerbl, 2023-11-09:
Hi everyone,

We have just released AlDente 1.23 which adds this functionality. To activate the battery percentage in the popover, open AlDente settings/Appearance/Popup Window and enable "Show Percentage". To get AlDente 1.23 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Cheers,
Matthias

closed 2023-11-09 by MatthiasKerbl