AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1030: Stuck at install helper app after macos update

andrhlvk, 2023-08-28:
**Describe the bug**
After updating macOS version to the latest 13.5.1, when I try to launch Aldente, the dialog to install helper application appears. When I click "Install Helper" I'm getting stuck at the install helper dialog window with a loading cursor in an infinite loop. The modal is not clickable in any way and never goes away unless I restart the computer, it is also above all the other windows. I have tried reinstalling the app and it did not help.

**To Reproduce**
Steps to reproduce the behavior:
1. Update macOS to 13.5.1
2. Launch Aldente
3. Click "Install Helper" in the launch dialog
4. The app is frozen

**Expected behavior**
The app to be launched

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro M1 pro 14 inch 2021 (MKGP3)
 - macOS version: 13.5.1
 - AlDente Pro or AlDente Free? Free
 - AlDente Version? 1.22.2
 - List of all enabled/disabled settings in AlDente:
 automatic updates are off, launch on boot is off

**Additional context**
Another interesting point is that my friend has the exact same setup (same aldente version, same mac, same macOS version) and everything's ok for him.

MatthiasKerbl, 2023-08-28:
Hi @andrhlvk,

Did you try to manually uninstall the helper application following our guide [How to uninstall AlDente](https://apphousekitchen.com/how-to-uninstall-aldente-pro/)

Best,
Matthias

andrhlvk, 2023-08-28:
> Hi @andrhlvk,
> 
> Did you try to manually uninstall the helper application following our guide [How to uninstall AlDente](https://apphousekitchen.com/how-to-uninstall-aldente-pro/)
> 
> Best, Matthias

Hello! Unfortunately I can't get to that remove option screen since the app isn't laucnhing past Install Helper state. I did try to just remove it from the finder and reinstalling though - did not help.

andrhlvk, 2023-08-28:
> > Hi @andrhlvk,
> > Did you try to manually uninstall the helper application following our guide [How to uninstall AlDente](https://apphousekitchen.com/how-to-uninstall-aldente-pro/)
> > Best, Matthias
> 
> Hello! Unfortunately I can't get to that remove option screen since the app isn't laucnhing past Install Helper state. I did try to just remove it from the finder and reinstalling with removing associated files though - did not help.

Apparently, I did not restart my pc according to the [unisntall guide](https://apphousekitchen.com/how-to-uninstall-aldente-pro/). 

Redid the whole thing once again, without the in-app remove option and the app seems to work just fine now.

Thanks!

closed 2023-08-29 by MatthiasKerbl

wangxiang4, 2025-09-22:
However, it is not recommended to directly modify the ```/Library/LaunchDaemons/com.apphousekitchen.aldente-pro.helper.plist``` file because re-installing the helper will cause the changes to be invalid. It’s better to create an agent that will start the ```com.apphousekitchen.aldente-pro.helper``` background process after the user enters the desktop. For more details, refer to the documentation: [macos-boot-software-invalid-issue](https://github.com/wangxiang4/macos-boot-software-invalid-issue#2macos-%E9%87%8D%E5%90%AF%E5%90%8E-aldente-%E9%9C%80%E8%A6%81%E5%AE%89%E8%A3%85-helper)