AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1031: Al Dente Pro Email Was Never Sent

kindrdghst, 2023-08-29:
Greetings,
I just purchased the pro version of Al Dente but never received the email.  Payment for $22.10 was sent 8/28/23 @ 1933 PST. 

Please advise!

Thanks,
Patrick

MatthiasKerbl, 2023-08-29:
Hi @kindrdghst,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Best,
Matthias

closed 2023-08-29 by MatthiasKerbl

kindrdghst, 2023-09-01:
Hey Mathias,
Can you kindly reach out to the Paddle team?  They are slow to respond and the money has already been taken from my bank account.

Thanks,
Patrick