AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1032: Mac does not wake up in clamshell mode

tradeqvest, 2023-09-03:
**Describe the bug**
When pressing on my keyboard or trackpad (connected via bluetooth), the Mac in clamshell mode connected via single USB-C to my monitor does not wake up if AlDente is active.

**To Reproduce**
Steps to reproduce the behavior:
1. Lock the Mac in clamshell mode, monitor goes black
2. Try to wake up Mac by pressing on keyboard/trackpad

**Expected behavior**
Mac should wake up and show the login window

**MacBook (please complete the following information):**
 - MacBook Air (M2, 2022)
 - macOS 13.5.1 (22G90) 
 - AlDente Pro 1.22.2

MatthiasKerbl, 2023-12-19:
Hi @tradeqvest,

First of all sorry for the late reply.
Unfortunately, I am unable to reproduce this issue.
Do you still experience this issue with our newest version?
What are your settings in AlDente?
Which charging state are you in before you close the lid? (charging, charging paused, discharging?)
Also, can you generate and share a debug file when this happens?

Best,
Matthias

Sandy-Pearl, 2024-01-02:
Hi, 

I'm having a similar issue. I'm running AlDente Pro 1.24 with a 2021 16-inch MacBook Pro M1 Max running macOS Sonoma 14.2.1 (23C71). The issue only appears when calibration is active, and my MacBook is discharging. 

I'm able to workaround this by unplugging my MacBook from my dock (which connects to my two monitors) and plugging it back in. I can confirm that both my mouse and keyboard are alive and connected to my Mac, but the displays remain black.

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes improvements for this scenario.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.
Please let us know if it works for you.

Best,
Matthias

rchrdyan, 2024-08-15:
I am running the latest AlDente 1.27.3 on Sonoma 14.6.1 (23G93) and have same 16-inch M1 Max setup as @Sandy-Pearl.
But I still experience exactly the same issue during every calibration.
@MatthiasKerbl

rchrdyan, 2024-09-16:
Still the same issue with the latest 1.28.2. Unable to wake up external monitor during discharging. @MatthiasKerbl

MatthiasKerbl, 2024-09-16:
Hi @rchrdyan,

Thank you for the continuous updates on this. We will try to fix this with the next updates.

Best,
Matthias

MatthiasKerbl, 2025-07-14:
Hi everyone,

We have just released AlDente 1.34.1, which includes a fix for this issue. To get AlDente 1.34.1, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-07-14 by MatthiasKerbl

mattdgold, 2025-07-29:
I'm still having this issue and I'm on 1.34.1.