AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1033: Macbook not charging

Su1cik, 2023-09-04:
After using free version of Aldente for more than a year without a problem. 75% limit that I had setup stopped working and after few days, macbook did not want to charge properly. When I plugged in the charger screen started flickering every second, because charging was pausing and starting every second. I uninstalled Aldente, made a SMC reset and everything seems to be working all right. 

Is it a known bug? I am little bit scared to install the program again, because I don't want to damage my computer. 

Macbook Air M1 2020, Mac OS 13.5.1

Thank you for your help.

ryanlalchand, 2023-09-23:
I was having the same problem on my MacBook Pro 16' 2019 on Ventura 13.6. Thanks to your comment about the SMC reset, it's charging now! I wish you the best of luck.

MatthiasKerbl, 2023-10-19:
Hi @Su1cik,

This might have been caused by a rare bug of AlDente 1.22.2. However, it has been fixed in AlDente 1.22.3.

Best,
Matthias