AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1034: Ignore heat protection when battery < 20%

Glydric, 2023-09-05:
**Describe the solution you'd like**
I would have a feature that allows the app to ignore the heat protection when the battery is under 20%, if it is hot, when I'm using the device hardly and i don't watch the battery percentage It might lower at a level where the macbook shut down itself. This lead to another problem, the battery drop under 20%, this is bad.

**Describe alternatives you've considered**
it might be implemented as two temperature limits, the one that we have now and a new temperature limit for when the battery is under 20%
For example i'm working in a heavy environment with a very hard task, the battery for some reason (i don't know exactly) drop to 10%, i see that the battery temperature was 1º more than heat protection limit, but this is not a problem, i prefer keep the battery at 20% instead of leading the device to shutdown, so when the battery is under 20% I want that the temperature limit is for example 38º

MatthiasKerbl, 2023-09-09:
Hi @Glydric,

Thank you for the suggestion.
I am not sure if I understood it correctly but why does your battery discharge when your MacBook is plugged in?

Best,
Matthias

Glydric, 2023-09-09:
Because the battery reaches the heat limit and aldente stop to charge the battery, on a long session this might lead the battery to completely discharge

dreit-p, 2023-09-09:
For example, playing a game takes a lot of energy for a long time, and sometimes takes more than the charger can give. So the laptop will discharge slowly

MatthiasKerbl, 2023-11-15:
Hi there,

In that case, the solution would be to use a stronger power adapter or deactivate Heat Protection. I think ignoring Heat Protection in general when the SoC is below 20% is counterintuitive and not the goal of the Heat Protection feature. Furthermore, it would just postpone the MacBook running out of battery in your situation and not avoid it completely.

Best,
Matthias

Glydric, 2023-11-19:
The stronger power adapter is not the problem, because AlDente blocks the charge also if device is down 20% and it can shutdown completely in a **high demanding task**. A solution could be an option to ignore temperature if battery percentage is down a threshold like 20-25%

MatthiasKerbl, 2024-01-24:
> The stronger power adapter is not the problem, because AlDente blocks the charge also if device is down 20% and it can shutdown completely in a **high demanding task**. A solution could be an option to ignore temperature if battery percentage is down a threshold like 20-25%

Hi @Glydric,

In my opinion, a strong enough power adapter would solve this issue. If the power adapter has enough power to always deliver enough energy to the MacBook, the battery would not discharge at all even if charging is paused due to Heat Protection. Again, as described above, we don't think turning off Heat Protection below 20% is a good solution.

Best,
Matthias