AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1036: When uncharging manually, macbook doesn't go to sleep even if disabled

ClementGre, 2023-09-09:
**Describe the bug**
When having the battery limit under the current charging level, and clicking on Discharge manually. And with all charging and sleep options disabled. The first time I close my macbook, it goes to sleep well, but when I try a second time it doesn't go to sleep (sometimes the bug occurs the third time or even after four or five tries). After this, the screen will never go off again except if you disable discharge or reach charge limit.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable all charging and sleep options as followed:
[image]

2. Put the charge limit at a percentage lower that the current charge
3. Click on the Discharge button :
[image]

4. Try to close and reopen the lid multiple times. After two or three try, the screen should stay always on, never going to sleep.

**Expected behavior**
The screen shoud be off and the mac should go to sleep as configured in the charge settings

**MacBook (please complete the following information):**
 - MacBook model: 2021 Macbook Pro M1 Pro
 - macOS version: Ventura 13.5.2
 - AlDente Pro, version 1.22.2
 - List of all enabled/disabled settings in AlDente:
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-11-10:
Hi @ClementGre,

Unfortunately, I am unable to reproduce this issue. Do you still experience this issue with our newest version AlDente 1.23?
Also, could you share a debug file when this happens again? [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/)

Best,
Matthias

ClementGre, 2023-11-11:
Hi,
I haven't noticed this issue these last few weeks, and I couldn't get to reproduce the issue with the latest version. I guess it was linked to the fact that turning off the "Disable Sleep until Charge Limit" did not have immediate effect.
But now this seems to be resolved (or maybe it was just a temporary bug of my Macbook).
Thank you !
Clément

closed 2023-11-11 by ClementGre