AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1037: Charge limit setting resets after right click

Volodymyr-13, 2023-09-14:
I've set the charge limit to 81% in the settings, with the sailing mode set to 5%. However, there's a persistent bug that occurs all the time. For instance, when the battery level is at 77%, the battery discharges slowly, indicating that the sailing mode is active and it's not charging.

Now, here's where the issue arises. I accidentally right-clicked on the aldente icon while intending to click on another app. After this, the charging process started, which I didn't want. So, I right-clicked again and saw that the charging was paused. However, the charge limit had been reset to 77%, and the sailing mode dropped to 72%. This means my settings were reset.

Even if I manually change the charge limit back to 81%, it starts charging again. Why does this happen? I don't want it to disrupt the sailing cycle. What I really want is for my settings and sailing cycle not to reset after an accidental right-click.

I've already requested to disable this right-click feature because it doesn't make sense for me. My MacBook has been stationary for two years, and I only took it with me for one day. I just want it to maintain an 81% charge, why not 80%? bcs of this bug: https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/678

[image]

MatthiasKerbl, 2024-04-12:
Hi @Volodymyr-13 ,

We have just released AlDente 1.26.1 which has a new setting that allows you to customize the Right Click behavior or turn it off completely.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl

Volodymyr-13, 2024-04-12:
Hi @MatthiasKerbl but pls add opition - Right click = same as Click, so I just need a menu, nothing else.

MatthiasKerbl, 2024-04-12:
Hi @Volodymyr-13,

Thank you for your feedback.
In my opinion, having the Right Click do the same as the "normal click" is quite redundant. Therefore, we will not implement it.

Best,
Matthias

Volodymyr-13, 2024-04-12:
@MatthiasKerbl I know you have opinion, since some features you totally try to ignore first and I want to prove to you that it is absolutely wrong.
Left -> Right clicks on a video: 

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/3391456/4e87f86d-d397-46c3-84e1-46f510a86620

vincentpun, 2024-04-25:
@MatthiasKerbl Can I ask you to please reconsider this? As @Volodymyr-13 showed in the video above, practically right clicking any macOS menubar icons behave the same as left clicking. (This is also true of the Wi-Fi icon before Big Sur.)

> Hi @Volodymyr-13,
> 
> Thank you for your feedback. In my opinion, having the Right Click do the same as the "normal click" is quite redundant. Therefore, we will not implement it.
> 
> Best, Matthias

MatthiasKerbl, 2024-06-18:
Hi everyone,

We have just released AlDente 1.27 which now also has a setting to mirror the "left click" when "right clicking" on the menubar icon.

Best,
Matthias

vincentpun, 2024-06-19:
Thanks for reconsidering and implementing this @MatthiasKerbl!!