AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1038: Simply does not work on Macbook Pro M2 Ventura

schteev, 2023-09-16:
Installed.. turned off apple battery optimization. installed helper app..set charge limit to 80%.. .it charged to 100%.
Uninstalled everything.. did it all again. same issue.
Uninstalled everything.. did it all again. same issue.
Uninstalled everything.. did it all again. same issue.

Wasted 2 hours of my day.. and I almost bought Pro right away. 

Complete junk.

DamonHD, 2023-09-16:
It works for me and plenty of others.  Why be so rude?

schteev, 2023-09-16:
Rude?
Whats rude is selling a product that isn't working properly - it wastes poples time. 
Then they are expected to come and troubleshott.. or endlessly uninstal / reinstal until they get lucky or give up?

I read a load of Macforums / Github etc posts from people who just gave up unistlling. reinstalling etc.. 

I'm one of those people who don't have time to mess around with something thats not 100%.
There are other 100% free solutions that I might expect not to work.. but this is a commercial product.. that doesn;t work out of the box.

closed 2023-09-16 by schteev