AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1040: Top Up doesn't have priority over macOS's automatic battery preservation

lumap, 2023-09-17:
**Describe the bug**
The Top Up feature doesn't have priority over macOS's built-in feature to preserve the battery

**To Reproduce**
Steps to reproduce the behavior:
1. Have AlDente Pro
2. Have your MacBook charging
3. Try to make macOS's automatic "Power will come from the power adapter, this MacBook will be charged at XX:00AM" behavior happen
4. Try to use Top Up while macOS has this mode activated

**Expected behavior**
The MacBook charges to 100% and macOS's battery preservation feature is disabled

**Screenshots**
The MacBook begins to charge, but macOS's feature kicks in after a few seconds, preventing it from charging

**MacBook (please complete the following information):**
 - Macbook Air (M1)
 - macOS Sonoma 14.0 RC
 - AlDente Pro 1.22.2

**Additional context**
[image]

MatthiasKerbl, 2023-09-18:
Hi @lumap,

Unfortunately, it is not possible to use Apple's Optimized Battery Charging and AlDente at the same time without any issues such as the one you encountered. Therefore, we are recommending everyone who wants to use AlDente should disable Optimized Battery Charging in macOS settings (Settings/Battery/Battery Health/"i").

Best,

Matthias

lumap, 2023-09-18:
Good to know, thanks!

closed 2023-09-18 by lumap