AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1041: Having battery percentage be in the bar icon

maxwowo, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
Hey there! Below is how I've configured the appearance of AlDente. You'll notice that although it's able to display both the battery percentage and the status of the charge, they take up quite a bit of bar real estate.

[image]

**Describe the solution you'd like**
Would it be possible to implement an option to have the percentage be in the icon alongside the current status symbol? Something like what the iPhone has would be ideal.

[image]

**Describe alternatives you've considered**
I have not yet found a suitable alternative.

**Additional context**
N/A

MatthiasKerbl, 2023-09-20:
Hi @maxwowo,

Good idea. We will try to add this option in one of the future updates.

Best,
Matthias

arafatx, 2024-11-16:
Hello, I was about to share this idea on the discussion page, but then I saw this post. It would be great to have the percentage integrated inside the battery icon to save space and make it look cleaner especially for crowded menu bars on Mac.

MatthiasKerbl, 2025-02-21:
Hi Everyone,

We have just released AlDente 1.31, adding an iOS Native menubar icon style.

Best,
Matthias

closed 2025-02-21 by MatthiasKerbl

mrzzmrzz, 2025-02-22:
> Hi Everyone,
> 
> We have just released AlDente 1.31, adding an iOS Native menubar icon style.
> 
> Best, Matthias

Hello, I noticed that there is a black line in the style of the native iOS menu bar icons. I'm wondering if this is a bug? 

[image]

MatthiasKerbl, 2025-02-24:
> > Hi Everyone,
> > We have just released AlDente 1.31, adding an iOS Native menubar icon style.
> > Best, Matthias
> 
> Hello, I noticed that there is a black line in the style of the native iOS menu bar icons. I'm wondering if this is a bug?
> 
> <img alt="Image" width="60" src="https://private-user-images.githubusercontent.com/36831062/415910371-e257f214-1a32-4412-910f-65515a3a1e1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzMTUzMzgsIm5iZiI6MTc0MDMxNTAzOCwicGF0aCI6Ii8zNjgzMTA2Mi80MTU5MTAzNzEtZTI1N2YyMTQtMWEzMi00NDEyLTkxMGYtNjU1MTVhM2ExZTFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjIzVDEyNTAzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlZjRlZjhmODFmMTBjMjE3MTQ5YjU1Yjc4ZmQxZmUyYzVkNjUwMTU1MTc4YjM0ZGViOGYxOTczMmFlZjYxZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9Ab43ZQG58CsNAXzXHSP21SzBzRSznKhTgaBhR6Rq58">

Hi @mrzzmrzz,

Thank you for letting us know. We will look into this.

Best,
Matthias

mrzzmrzz, 2025-03-17:
[image]

It seems that the resolution of the lightning charging icon inside is quite low, not entirely sure if that's normal?

mrzzmrzz, 2025-03-17:
[image]

This charging icon also appears to have a similar issue with low resolution.

MatthiasKerbl, 2025-03-18:
Hi @mrzzmrzz,

Thanks for letting us know, we will look into it.

Best,
Matthias