AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1042: No automatic percent localisation

bitigchi, 2023-09-20:
The percent format looks like it's hardcoded to 100%. However, there are many languages that uses different percent sign positions or even different percent signs. To remedy this, `NumberFormatter()` or SwiftUI's `.formatted(.percent)` should be used.

bitigchi, 2024-03-01:
Any chance this could be given a priority? Looks very unprofessional to eyes...

MatthiasKerbl, 2024-03-11:
Hi @bitigchi,

In which language do you experience this problem?

Best,
Matthias

bitigchi, 2024-03-11:
Hello @MatthiasKerbl,

Thanks for the reply. I use it in Turkish, but [all these languages](https://en.wikipedia.org/wiki/Percent_sign#Form_and_spacing) would benefit from this fix.

Best,
Emir

MatthiasKerbl, 2024-03-11:
Hi @bitigchi,

Thanks for letting us know, we will try to add it in one of the upcoming updates.

Best,
Matthias

bitigchi, 2025-04-03:
Any update on this? Kind of surprised that such a basic thing is overlooked...

MatthiasKerbl, 2026-10-19:
Hi @bitigchi,

Sorry this took so long. AlDente 1.40 Beta 1 now formats percentages according to your language settings.

AlDente 1.40 Beta 1 is currently available for AlDente Pro (Settings → gear icon at the bottom left → "Receive Beta Updates" → "Check for Updates"), and the final release for everyone will follow soon. We'll close this issue once 1.40 is out.

Best,
Matthias

bitigchi, 2026-10-20:
> Hi @bitigchi,
>
> Sorry this took so long. AlDente 1.40 Beta 1 now formats percentages according to your language settings.
>
> AlDente 1.40 Beta 1 is currently available for AlDente Pro (Settings → gear [...]

Thank you very much! Better late than never. 😊