AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1044: Please add a quick toggle for low power mode on the menu bar icon.

CollinsVakayil, 2023-09-20:
Currently, the only way to activate Low Power Mode on macOS that I'm aware of is by going into the settings > battery > toggle low power mode. Which is a lot of clicks. It would be great to have a quick toggle feature to turn on low power mode from the menu bar.

DamonHD, 2023-09-20:
I would like toi be able to control it from the AlDente scheduler.

For now I adjust it from cron autmatically as described here:

https://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html#lowpowermode

Rgds

Damon

MatthiasKerbl, 2023-09-21:
Hi @CollinsVakayil,

Good idea, there is already a feature request for this open #603. Therefore I will close this thread and post any updates in #603.

Best,
Matthias

closed 2023-09-21 by MatthiasKerbl