AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1045: Macbook Air M2 not charging

bvidovic1, 2023-09-21:
After 6 months of actively using AlDente Pro, one day it started to behaving strange. Battery is not being charged even tough battery level is below charge limit. I  followed your guideline and uninstalled AlDente from my system. However, even after that my battery is not being charged anymore. I tried turning it off, close the lid wait, rebooted many times - no help.

I am using my MagSafe charging cable that I have been using since I bought my laptop.

Is there any recovery procedure?

**MacBook (please complete the following information):**
 - MacBook model: Macbook Air M2 2022
 - macOS version: Ventura 13.5.2
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.22.2
 - List of all enabled/disabled settings in AlDente: Currently I disabled everything, to test, but nothing worked.

ryanlalchand, 2023-09-23:
I'm having the same problem on my MacBook Pro 16' 2019 on Ventura 13.6.

phocean, 2023-09-27:
Same issue here, Mac OS Sonoma, Mac M1.

tswistak, 2023-10-02:
I have the same problem with Macbook Pro with M2 Max. Despite the charge limit set to 70%, the app doesn't charge the battery when the value is lower.

marcoluzi, 2023-10-02:
Same problem here on my MacBook Pro (2017) and MacBook Pro (2021) both running macOS 13.6.

bvidovic1, 2023-10-02:
Did any of you guys perform an update of OS whether minor or major version? I did some investigation and it seems like 1 week after I did the upgrade shitty behavior started.

I uninstalled AlDente (which I paid btw) by following their docs and reinstalled macOS (you can reinstall without erasing the data and apps). I reinstalled it hoping it would reset firmware settings and any other potential settings that AlDente impacted. After that, my MagSafe started charging again.

Since this will probably happen after each upgrade, I am very reluctant to install AlDente onto my MacBook again because I am afraid I will need to go through this process after each OS upgrade.

marcoluzi, 2023-10-04:
@bvidovic1 
I think it happened after the update to 13.6, but I'm not sure. 

I am also not quite sure wether it is an AlDente Problem or something else.
I can't charge via USB C. It works sometimes with MagSafe but not always, and if it works it is charging very slowly.
Problem persists when AlDente is not running.

MatthiasKerbl, 2024-02-20:
Hi @bvidovic1,

Do you still experience this issue?
Also, we have just released AlDente 1.25.2 which has an updated debug file. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

MatthiasKerbl, 2024-02-26:
Hi @bvidovic1,

I am sorry you had such a bad experience and I understand that you are hesitant about installing AlDente again.
If you want to, you can send an email to our official support, and we will try to figure something out for you.

Best,
Matthias

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes a fix for this issue.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl

bvidovic1, 2024-04-14:
Thanks for information. I am getting new Macbook soon, so I will install AlDente on my old one and try it again :)