AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1046: Sonoma 14.0 (23A339) Menubar Icon sometimes fails to show up.

pnelego, 2023-09-21:
**Describe the bug**
On **BETA** Sonoma, the icon on the menubar occasionally doesn't show up, even though AlDente is running and handing charge levels. Here are the settings I'm using otherwise regarding appearance. 

[image]

Work-around:
Quitting the app via the Activity Monitor and restarting rectifies the issue.

Note: Am aware that this issue is potentially due to MacOS beta being weird, but with the official release approaching, I figured you might have wanted to get ahead on it.

**To Reproduce**
1. Set up AlDente with the settings listed above
2. Shut Down and restart computer
3. AlDente will start in the "background" (without a dock icon, as expected), however menubar icon isn't present.

**Expected behavior**
Expected menubar icon to load

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model A2485
 - macOS version 14.0 (23A339 RC)
 - AlDente Pro 
 - AlDente Version 1.22.2
 - List of all enabled/disabled settings in AlDente:
 
[image]
[image]
[image]

MatthiasKerbl, 2023-09-22:
Hi @pnelego,

Thank you for reporting this issue. Unfortunately, we have not been able to reproduce this issue so far.
If this issue happens again, try to enable the settings "Only show AlDente Icon" and let us know if the icon appeared in the menubar.

Best,
Matthias

derbalkon, 2023-09-27:
I have the same issue after setting the menubar icon to Show Percentage with macOS Native. But it would show up if I force quit the app and restart it.

> If this issue happens again, try to enable the settings "Only show AlDente Icon" and let us know if the icon appeared in the menubar.

Also, with the setting "Only show AlDente Icon" enabled, the icon will not disappear after restarting the system.

pnelego, 2023-09-27:
> I have the same issue after setting the menubar icon to Show Percentage with macOS Native. But it would show up if I force quit the app and restart it.
> 
> > If this issue happens again, try to enable the settings "Only show AlDente Icon" and let us know if the icon appeared in the menubar.
> 

I think the issue may no longer be present in 23B5046f, I'm curious if you're seeing the same thing.

dalemugford, 2023-09-28:
I have the same issue— Al Dente Pro 1.22.2 on MacOS Sonoma Version 14.0 (23A344)

MatthiasKerbl, 2023-10-01:
Hi everyone,

Short update about this bug from our side. We are currently still investigating this bug. Unfortunately, we are not able to reproduce it at this point. Please let us know if you found a way to consistently reproduce this issue. Theoretically, if you activate "Only AlDente Icon" in the Appearance settings, this bug should not occur.

Best,
Matthias

pnelego, 2023-10-01:
Im starting to wonder if it was a MacOS bug manifesting. On the latest beta for macOS, and i've not had the issue a single time.

MatthiasKerbl, 2023-10-01:
> Im starting to wonder if it was a MacOS bug manifesting. On the latest beta for macOS, and i've not had the issue a single time.

Thanks for reporting back @pnelego. This could of course be the case.

Best,
Matthias

jimococo, 2023-10-02:
在macOS14的正式版上，仍然会出现这个问题。复现的方法是，关机，然后再开机，而不是重新启动。此问题就会出现。This problem still occurs on the official version of macOS14. The way to reproduce is to shut down and then back on instead of restarting. This problem will occur.

OneSman7, 2023-10-06:
Just noticed this bug. Updated to Sonoma 2 days ago. Shutting down my system affter each workday. Yesterday everything was ok. Today the icon did not show up. I rebooted and still no icon. After I found this thread I restarted Al Dente from Activity Monitor to fix the issue. I prefer to use icon with Al Dente status.

closed 2023-10-10 by pnelego

Slam-Thompson, 2026-03-09:
Same on M1pro with macOS Tahoe 26.3.1. Whatever I did, the menu bar icon cannot go back, even if I uninstall and reinstall it. How to find the menu bar icon? Now there is no way to control the charging limit.

MatthiasKerbl, 2026-03-09:
> Same on M1pro with macOS Tahoe 26.3.1. Whatever I did, the menu bar icon cannot go back, even if I uninstall and reinstall it. How to find the menu bar icon? Now there is no way to control the charging limit.

Hi @Slam-Thompson,

Please try the following:

1. Open macOS settings
2. Go to Menu Bar
3. Scroll down to "Allow in the Menu Bar"
4. Enable the slider next to AlDente

Best,
Matthias