AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1047: Mac do not sleep while Al Dente Pro is running

spsisys, 2023-09-23:
**Describe the bug**
Read the title. There are many requests about this problem from 2022.

**To Reproduce**
Just open the app. Sleep menu option is disabled and sleep & sleep displays doesn't work any more.

**Expected behavior**
I think it's obvious.

**Screenshots**
Al Dentro Pro Running:
[image]
[image]

Al Dente Pro Closed:
The system go to sleep properly. 
[image]

**MacBook 
[image]
ase complete the following information):**

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-09-25:
Hi @spsisys,

This is actually the intended behavior of AlDente Pro when the Discharge feature is active to ensure discharge works in all scenarios such as the clamshell mode for example. Since you have Automatic Discharge enabled and you are currently discharging, sleep is disabled by AlDente.
Also, when discharging for the first time, a short explanation should appear explaining this scenario.
[image]

Best,
Matthias

spsisys, 2023-10-02:
@MatthiasKerbl Thanks for your answer. All clear! But I need the sleep mode active even when the battery is discharged. I understand that it's a so easy step to cancel Discharge Mode, but it's not my workflow because I've to leave my workplace often, and the app is installed to prevent the battery from further charging. Nevertheless, I'll check the new workflow before uninstall the app. Cheers!

closed 2023-11-10 by MatthiasKerbl