AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1048: Sonoma 14.0 RC2 - AlDente Not Limiting Charge

kaikozlov, 2023-09-24:
**Summary**
I am on the latest beta release of Sonoma, and I have noticed that AlDente has not been functioning properly to limit charge or enable sailing mode.

**To Reproduce**
1. Install the latest Sonoma beta. 
2. Install AlDente.
3. Charge Mac

**Expected Behavior**
I expect that my battery should be limited at 80% and sailing mode should be activated.

**Screenshots**
[image]

**MacBook Information**
- MacBookPro18,4
- macOS 14.0 (23A344)
- AlDente Pro 1.22.2
- Settings Enabled: Automatic Updates, Launch at Login, Hardware Battery Percentage, Sailing Mode (10%), Show Percentage

MatthiasKerbl, 2023-09-25:
Hi @kaikozlov,

Thank you for sharing this issue. Sonoma is not yet officially supported and we are still working on eliminating all compatibility issues.
Could you try to activate the "Stop Charging when App Closed" settings and check if the Charge Limiter works afterward?
Furthermore, it would be helpful if you could share a debug file from when this issue is occurring. ([How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/))

Best,
Matthias

kaikozlov, 2023-09-25:
Hi @MatthiasKerbl,

Enabling the "Stop Charging when App Closed" setting seems to restore the charge-limiting functionality. Attached below is the requested debug file.

Thanks,
Kai

[aldentepro-debug-2023-09-25-08:44:56.pasta.txt](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/12718325/aldentepro-debug-2023-09-25-08.44.56.pasta.txt)

LKuech, 2023-09-27:
I experienced the same issue with the final version of Sonoma that was released yesterday. 
Interesting to note is that the MagSafe-LED is green for some seconds and then becoming orange for a second or so. This is happening as soon as the charge limit is passed and the battery is charged over the configured limited. The icon in the menu bar is switching from the "pause" icon to the "plus" icon in the same moment for a briefly as well. 

It seems like the MAG-Safe behavior is not changing when "Stop Charging when App Closed" is enabled.

gusost, 2023-09-27:
Would be really good to get a time estimate for Sonoma support.
I am an AlDente Pro user and this is what is currently stopping me from updating to Sonoma

LKuech, 2023-09-27:
> Would be really good to get a time estimate for Sonoma support. I am an AlDente Pro user and this is what is currently stopping me from updating to Sonoma

Yep. Usually, beta and release candidate phases do exist to allow developers to find compatibility issues before their users do 😜

LKuech, 2023-09-27:
Okay. Good point.

I my case it is a Macbook Pro M1 2021.

phocean, 2023-09-27:
There were many betas and yet AlDente, which plays a very critical function on the computer, is not ready at all.
The current behavior on Sonoma is an absolute mess : not charging, then starting to charge but stops early, etc.
My Macbook M1 is unusable because of that.

rolleur, 2023-09-28:
After updating to Sonoma on the release date Al Dente didn't seem to be working since the battery percentage wasn't coming down (it had reached 100% during the updating process).

I was initially frustrated, but reinstalled the helper application, turned on "Stop Charging When App Closed", restarted, and unhooked power until the charge dropped into my normal sailing range.  Then I reconnected to power and everything has been fine since.  I'm not sure which step got it working again, but everything is great for my MacBook/Sonoma/Al Dente combo again.

MacBook Information

MacBookAir10,1
macOS 14.0 (23A344)
AlDente Pro 1.22.2
Settings Enabled: Automatic Updates, Launch at Login, Hardware Battery Percentage, Sailing Mode (7%), Show Percentage, Heat Protection, Stop Charging When App Closed

LKuech, 2023-09-28:
Well... Kind of the same story on my end. Right now, after several restarts etc. It is working without the weird behavior mentioned above. I do even have the option "Stop Charging when App Closed" disable again (like I had before).
Everything seems to work so far... Stop charging at 75% and sailing down to 65% ... 
Let's wait some charge cycles.

martin-bryndza, 2023-09-30:
Sonoma has been out for quite a while now and there still isn't any update from Al Dente.  On their web page they still claim that AlDente Pro is currently only supported on **MacOS 11 Big Sur or later**. Is the app development dead?

MatthiasKerbl, 2023-10-01:
Hi everyone,

Short update from our side. We have been testing AlDente on the macOS Sonoma betas for months and have not encountered any issues on our devices. Due to the inconsistent nature of this bug, we were only made aware of the bug mentioned in this thread by one of our users a while ago and implemented a fix in the next update of AlDente. Unfortunately, due to bigger changes in the background of the next update of AlDente, development and beta testing from test users took longer this time than expected. However, testing is looking good and the update will most likely be released at the end of next week.

We are sorry for any inconvenience this may have caused and will try our best to avoid such issues in the future.

Best,
Matthias

MasterOfMouse, 2023-10-01:
> Hi everyone,
>
> Short update from our side. We have been testing AlDente on the macOS Sonoma betas for months and have not encountered any issues on our devices. Due to the inconsistent nature of this [...]

Hi, glad to see Al Dente will be back in action soon. I updated from 13.5.1 to 14.0 and Al Dente had become so ingrained into my system that I forgot about it! Great piece of software and I hope that you continue to prioritise 'software-lightness' and avoid feature creep. Nice that the program is around 16-17 MB? and it is much lighter than competing programs in size.

MatthiasKerbl, 2023-10-01:
> Hi, glad to see Al Dente will be back in action soon. I updated from 13.5.1 to 14.0 and Al Dente had become so ingrained into my system that I forgot about it! Great piece of software and I hope [...]

Hi @MasterOfMouse,

Thanks, that's much appreciated. We also try to keep AlDente as light as possible but always offer new features, improvements, and customizability.

Best,
Matthias

gusost, 2023-10-02:
Thanks for the timeline update, much appreciated!

juliangalardi, 2023-10-03:
Same issue here:

OS:  Sonoma 
Version: 14.0 (23A344)
Processor: Apple M1 Pro

joshjudgere, 2023-10-03:
Is there a beta we can use in the interim? Downgrading to Ventura till next week is an ordeal.

mojoba, 2023-10-03:
Is there a beta we can use till next week?

gexcube, 2023-10-04:
M2 air as well :) Thank you for fixing!

fabiotavares, 2023-10-04:
Air M1 here...
I also had problems after upgrading to sonoma.
The percentage did not appear on the taskbar.
I reinstalled the app and it appeared!
However, when restarting the Mac, despite the app being launched, the percentage again did not appear in the bar. But if you close the app and open it manually it appears.
So, when it starts automatically with Mac, I need to manually restart the app for it to work.

john41nt, 2023-10-05:
MacBook Air 13 M2, latest version of Al Dente Pro
When I noticed the battery charge limit was not working as expected after updating macOS 14 yesterday, it was charged to 94% already. I reinstalled the plugin, set battery to discharge (still connected), it was discharging for a few hours but went to charging again. I'm afraid the only change was I connected the system to a monitor (via USB-C with 90W PD). This action never impact battery charge limitation with macOS 13.

[image]

When I found the system was charged to 100% today, I set battery to discharging again but the system continued to switch charging/discharging every a few seconds. It looks fine after I rebooted the system. Will update for further status.

john41nt, 2023-10-06:
Update: The tool is working as expected till now. What I did yesterday is re-install the plugin and reboot.

MasterOfMouse, 2023-10-06:
> Update: The tool is working as expected till now. What I did yesterday is re-install the plugin and reboot.

Thanks, re-installing the 'helper' and restarting seems to have fixed the issue. However I have not left the machine with its lid closed yet.

LKuech, 2023-10-06:
> > Update: The tool is working as expected till now. What I did yesterday is re-install the plugin and reboot.
> 
> Thanks, re-installing the 'helper' and restarting seems to have fixed the issue. However I have not left the machine with its lid closed yet.

Yep. Had the same behavior. But after several days the issue reoccurs and a restart was needed again. So right now, it is not as reliable as before. Let's wait for the update next week.

belko2k, 2023-10-09:
i am having a problem when I try to discharge from 100% back to 80%. My external monitors immediately go to sleep when I press discharge button.

MatthiasKerbl, 2023-10-10:
Hi everyone,

We have just released AlDente 1.22.3 which includes a fix for this issue. To get AlDente 1.22.3 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Thank you for your patience.

Cheers,
Matthias