AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1052: external Display cannot be reactivated when using USB Type C connection

Bat-Maniac, 2023-09-28:
**Describe the bug**
My external Display cannot be reactivated when AlDente is running. I have to disconnect and reconnect the USB-C connection every time to get the monitor to reactivate after the display turns off (I also tested it with another monitor - with the same result).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect MacBook with USB Type C to an external Display
2. Wait until the display turns off after the specified time
3. Try reactivating the external display using the keyboard/mouse/trackpad
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook Air M1, 2020
 - macOS Sonoma 14.0
 - AlDente Pro
 - AlDente 1.22.2

MatthiasKerbl, 2023-09-29:
Hi @Bat-Maniac,

Thanks for sharing this bug and we are sorry you are experiencing this issue. Unfortunately, this behavior is rather inconsistent and we are currently not able to reproduce this issue on our devices.

Is your Sleep option in macOS greyed out while Discharge is active in clamshell mode?
[image]

Best,
Matthias

Bat-Maniac, 2023-09-29:
Hi @MatthiasKerbl,

thanks for your response.

Yes, sleep option is greyed out while on discharge. 

Maybe I need to connect my MacBook to the display via Displayport instead of USB C and then connect the MacBook to the charger for power.

Then I could simply turn off the monitor and the MacBook would still receive power.

I hope there will be a solution, otherwise the display will simply stay on as long as AlDente discharges the battery.

But thank you for this great app and for the time you invest in it.

mechantokun, 2025-05-13:
Hi,

Recent user of AlDente Pro since this year and I noticed a similar issue when using my macbookpro M4 in clamshell mode with an external monitor via Hdmi. I programmed some discharge along the day with your scheduler option (works nice) but I notice that when on discharge and letting my macbookpro inactive it will turn off the screen and the only way to turn my monitor back on is to unplugg my magsafe charger few seconds and plug it again. It wlll break my Scheduler discharge but I have no other choice.

So for now I will try the OP's found workaround : 
> 'turn display off on battery when inactive ' to 'never' in macOS lock screen settings.

If you could add an option in aldente that that we could enable to prevent this behavior it would be great (or show me where if it exists).

Anyway Big thumbs up for this app and Happy to be a Pro user.

MatthiasKerbl, 2025-07-14:
Hi everyone,

We have just released AlDente 1.34.1, which includes a fix for this issue. To get AlDente 1.34.1, you need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-07-14 by MatthiasKerbl