AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1055: Removal of the top arrow/indicator on the popover

Bat-Maniac, 2023-09-30:
**Describe the solution you'd like**
In my opinion the popover would look better if the arrow/indicator at the top could be removed. This would visually adapt better to macOS popup menus.
Maybe you could make the popover a little smaller.

[image]

**Additional context**
Here is an example popover with sliders wich looks more macOS like.

[image]
 _app: monitor control_

MatthiasKerbl, 2023-10-01:
Hi @Bat-Maniac,

Good idea. We are open to changing the design of the pop-over window in one of the next updates.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
Hi @Bat-Maniac,

We have just released AlDente 1.28.1 which includes a fix for this issue. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl