AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1057: [Feature Request] Option for battery temperature in Fahrenheit

evank21, 2023-10-01:
Would be nice to have an option to show the battery temperature in fahrenheit, using something like [this converter](https://www.npmjs.com/package/celsius) maybe. Can't find any open/closed issues or FAQs about this.

MatthiasKerbl, 2023-10-01:
Hi @evank21,

Good idea, that's definitely something we could add in a future update.

Best,
Matthias

MatthiasKerbl, 2024-02-09:
Hi @evank21 ,

We have just released AlDente 1.25 which will automatically switch to Celsius or Fahrenheit depending on the macOS setting. To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl

sooswastaken, 2024-12-03:
Hello, could I request that this be a separate setting from the operating system (could default to sync)? Thank you.

dvanburen, 2025-05-20:
Another vote for a user override. A simple C, F or "Follow System" option would be great.