AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1059: Bugs, problems and device killer

Bat-Maniac, 2023-10-01:
I've used AlDente (Pro) more often now because the promised features sound good. But in reality you can't really use any of them without sacrificing system services.

I mostly use my MacBook in clamshell mode with an external display connected via USB Type C, so I only need one cable to transfer the image and the MacBook is powered at the same time, which is very practical.

But the problems with AlDente begin with this setup.

The external display is no longer automatically deactivated after a set time. If you switch it off manually (active edges), it can no longer be activated using the mouse/keyboard/trackpad. Simply disconnecting and reconnecting the USB port helps here. However, tasks that are carried out by AlDente, such as calibrating the battery, are then aborted.
In order to turn off the display, I would have to turn off the device completely, which also results in the AlDente tasks being canceled because the power supply is then interrupted.
The fact that sleep mode no longer works with almost any AlDente feature is also absolutely unacceptable.
Not even the Screesaver is activated automatically when AlDente is running. This means I now have a burn-in on my display.
This means I have to spend a lot of money and buy a new display just to extend the life of the MacBook battery by a few days.

This is absolutely not a compromise. I say bye bye AlDente.

bvidovic1, 2023-10-02:
Same here.
Purchased the Pro version, and it was working very well on macOS 12, but after upgrading to 13 I started noticing all different sorts of issues. I managed to recover my Macbook finally and I am not planning on installing and using AlDente again because I am afraid my device will stop working when I need it most.

MatthiasKerbl, 2025-05-15:
Hi everyone,

I'm sorry to hear you're not satisfied with AlDente. When it comes to the issue of the display not turning off when AlDente has to disable sleep, we were finally able to improve on that, and AlDente is now able to fully mirror the macOS setting.

Best,
Matthias

closed 2025-05-15 by MatthiasKerbl