AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1061: The charging limit value cannot be changed in the "Set Charging Limit" of the scheduled task, sonoma 14.1 beta system

ccmilu, 2023-10-02:
**Describe the bug**
In the scheduled task, create a new "Set Charging Limit". At this time, I want to change the charging limit value to "90", so I press the delete key to delete the default value "80". However, once the value in the input box is Delete, aldente will immediately reset the contents of the input box in a rapid manner, causing me to be unable to enter any new numbers. sonoma 14.1 beta system

**To Reproduce**
System version: sonoma, 14.1beta
Steps to reproduce the behavior:
1. Go to "Schedule"
2. Click the plus sign below to add a task
3. Change the task type to "Set Charging Limit"
4. Click the input box for the charging limit value
5. Delete the default value
6. See error, you cannot change any value, aldente will reset the input box very quickly

**Expected behavior**
I just wish I could change the charge limit value in a mission normally, please, this bug is driving me crazy!

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model A2338
 - macOS version sonoma 14.1beta
 - AlDente Pro
 - AlDente Version 1.22.2
 - List of all enabled/disabled settings in AlDente:
     - sailing mode:on
     - heat protection:on
     - Stop charging when sleeping:on
     - others:off

**Additional context**
none

MatthiasKerbl, 2023-10-02:
Hi @ccmilu,

Thank you for reporting this issue. This will be fixed with the next update.

Best,
Matthias

MatthiasKerbl, 2023-10-10:
Hi @ccmilu,

We have just released AlDente 1.22.3 which includes a fix for this issue. To get AlDente 1.22.3 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Cheers,
Matthias

MatthiasKerbl, 2023-10-13:
Hi @ccmilu,

Since this bug is fixed now, I will close this issue.

Best,
Matthias