AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1062: Sailing mode not working properly since Sonoma Update

Richard-MD, 2023-10-03:
**Describe the bug**
The MacBook is going back and forth between battery and power adapter. When the MacBook is plugged into the power adapter, plugged into a monitor and closed the display turns off and on. The MagSafe indicator flashes between orange and green.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug MacBook into the power adapter
2. Plug MacBook into a monitor
3. Close MacBook
4. Once the MacBook has reached the % limit set by user it will reproduce the issue.

**Expected behavior**
Having the MacBook plugged into a power adapter and monitor with it closed had no issues before the Sonoma update.

**Screenshots**
See attached video recording of the issue.

**MacBook (please complete the following information):**
 - MacBook model MacBook Air M2, 2022
 - macOS version Sonoma 14.0
 - AlDente Pro
 - AlDente Version? - 1.22.2
 - List of all enabled/disabled settings in AlDente:
  - Automatic discharge
  - Sailing Mode 5%

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/128251837/80834a69-7e47-421e-a4c7-e192eefca0fb

mojoba, 2023-10-04:
May be related but even with AlDente off, the battery won't charge sometimes while plugged in even with 'Optimized Battery Charging' off. Not sure if this is a Sonoma bug or something else related to AlDente.

Strife97, 2023-10-04:
Sailing mode not working properly. Charge Limit to 75%, Sailing mode 10%. But after discharging to 70%, battery start charging again to charge limit when plug into the power adapter. [Macbook Air M1]

MatthiasKerbl, 2023-11-10:
Hi @Richard-MD,

This seems to be a known bug documented in this issue #768.

In short, this is how you should be able to fix it:

AlDente stopped working after the macOS update!

Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.

To get AlDente to work again, please try the following steps which have worked for many users before:

Unplug and discharge your MacBook to less than your current charge limit in AlDente
Turn off your MacBook and close the lid for a couple of minutes
Unplug the power brick from the outlet and wait a couple of minutes
Turn on your MacBook and plug it in again
Check if it stops charging at the charge limit set in AlDente
We have been aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Best,
Matthias

MatthiasKerbl, 2023-11-29:
Hi @Richard-MD,

Any update on this?

Best,
Matthias

closed 2023-12-18 by MatthiasKerbl