AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1063: Calibration not working in Sonoma

griffinmartin, 2023-10-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
- Start calibration
- Charge --> Discharge --> Charge 100% 
- Went away for 30 minutes
- Screen was black
- `Start Calibration` option is available and it is not calibrating

**Expected behavior**
- Calibration does a 1-hour hold and then automatically discharges.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - 2021 16" M1 Max
 - macOS version (https://support.apple.com/en-us/HT201260)
 - Sonoma 14.0
 - AlDente Pro or AlDente Free?
 - Pro
 - AlDente Version?
 - 1.22.2
 - List of all enabled/disabled settings in AlDente:
 - `Enabled` - stop charging when sleeping, automatically discharge, disable sleep until charge limit, lock screen when closed, hardware battery percentage, sailing mode, heat protection
 - `Disabled` - Stop charging when app closed

**Additional context**
Add any other context about the problem here.
- I just got this app yesterday; this is my first calibration attempt.

MatthiasKerbl, 2023-10-05:
Hi @griffinmartin,

Thank you for reporting this issue. Unfortunately, I am currently not able to reproduce this issue. Are you consistently able to trigger it?
I have a couple of questions to assess your situation more thoroughly:

1. Did you use an external display?
2. Did you use your MacBook with the lid closed/in clamshell mode?
3. Is the macOS Sleep icon greyed out during the Calibration Mode?

Also, please share a [debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) from the next time this is happening.

Best,
Matthias

goldjunge91, 2023-11-25:
Hi,
I have a similar Issue as @griffinmartin.
I'm testing AlDente Version:1.24 (47) today for the first time.
(Of course, Griffin Martin's issue has priority over mine.)

**I still use my MacBook Air when i start the Calibration, so maybe that's a problem?**

Model: Macbook Air M1  [Link](https://support.apple.com/kb/SP825?locale=de_DE)
MacOS:14.1.1 (23B81)
List of all enabled/disabled settings in AlDente:
**Enabled** - 
Stop charging when sleeping,
Stop charging when powered off/app closed,
Hardware Battery Percentage,
**Disabled** -
Automatic Discharge
Disable Sleep until Charge Limit

> 1. Did you use an external display?
**Yes**.

> 2. Did you use your MacBook with the lid closed/in clamshell mode?
**No.**

> 3. Is the macOS Sleep icon greyed out during the Calibration Mode?
**Yes.**

<details open>
  <summary>Screenshots</summary>

### Settings:

[image]

### Settings:

[image]

### Second Stage of Calibration:

[image]

</details>

<details open>
  <summary>log</summary>

 **First Try with restart and start of second time started**

[aldentepro-debug-2023-11-25-23:20:08.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13465404/aldentepro-debug-2023-11-25-23.20.08.pasta.zip)

**Second Try after Second Stage of Calibration (Discharging 15%)** 

[aldentepro-debug-2023-11-25-23:39:42.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13465406/aldentepro-debug-2023-11-25-23.39.42.pasta.zip)

</details>

andrewcrook, 2023-12-12:
I have just had the same issue 

- macOS 14.1.2
- Aldente Pro 1.24

- MBP M1 13” 2020
- External Monitor  LG 27UK850 display port over thunderbolt 3/ USB 4
- Apple Magic Keyboard bluetooth
- Logitech MX Master 3  via Logitech USB Unifying Receiver

1. clamshell mode
2. on AC mains (direct to mains)
3. Sleep greyed out 
4. left over night with display off
5. used today
6. calibration stage was on 1 hour hold
7. I turned off the display and left  for 2hrs
8. came back and calibration had ended, however, it didn’t have time to do the 100% discharge at the end so know that it didn’t complete.

andrewcrook, 2023-12-12:
ahh that’s interesting the calibration discharge has change from 100% to 85% 
when I set it back to my normal settings of limiting to 85%.

MatthiasKerbl, 2023-12-16:
Hi @andrewcrook,

> is the last discharge meant to be 100% or the original setting before starting calibration?

The Calibration Mode always safes the set charge limit from before the calibration starts and will at the end of the calibration discharge to this charge limit.

Best,
Matthias

closed 2024-01-23 by MatthiasKerbl