AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1064: Free Version - Unable to activate "Stop charging when sleeping" function - Sonoma

ademlabs, 2023-10-04:
**Describe the bug**
After the Sonoma update, AlDente Free doesn't allow me to activate the stop charging when sleeping option. This causes the battery to charge to 100% when sleep is triggered.

On Monterey this option was active and fully working.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Free
 - AlDente Version 1.22.2

MatthiasKerbl, 2023-10-05:
Hi @ademlabs,

This feature has always been exclusive to AlDente Pro and there have not been any changes to this from Ventura to Sonoma. Also, charging during Sleep is also paused in AlDente Free when the charge limit is reached before the MacBook goes into Sleep mode.

Best,
Matthias

ademlabs, 2023-10-05:
I'm not sure what changed, and my mac used to charge up to the limit I've set (e.g. 80%) and stop charging before going to sleep. Now it seems the computer will charge up 100% regardless. Both cases when I close the lid when still plugged in.

Maybe AlDente is not inhibiting sleep until charge limit is reached? Only started happening after Sonoma upgrade. I'll try uninstalling and reinstalling to see if it resolves.

ademlabs, 2023-10-07:
So, I reinstalled AlDente and it still isn't working properly.
I used the computer until battery was around 50%. I then plugged it in, then after a few mins I closed the lid. The sleeping wasn't inhibited apparently and the charge went up to 100%.
Not sure what is broken, here.

ademlabs, 2023-10-25:
AlDente still doesn't work correctly. I switched to using [battery](https://github.com/actuallymentor/battery) for the past 3 weeks and haven't had any issues with it overcharging my Mac. Seems that there is indeed something broken somehow with AlDente on Sonoma with my setup.

MatthiasKerbl, 2023-10-31:
> So, I reinstalled AlDente and it still isn't working properly. I used the computer until battery was around 50%. I then plugged it in, then after a few mins I closed the lid. The sleeping wasn't inhibited apparently and the charge went up to 100%. Not sure what is broken, here.

Hi @ademlabs,

"Disable Sleep until Charge Limit" is only available in AlDente Pro. Therefore, if your MacBook is still charging while it goes to sleep, it will continue charging to 100%.

Best,
Matthias

MatthiasKerbl, 2023-10-31:
> AlDente still doesn't work correctly. I switched to using [battery](https://github.com/actuallymentor/battery) for the past 3 weeks and haven't had any issues with it overcharging my Mac. Seems that there is indeed something broken somehow with AlDente on Sonoma with my setup.

What exactly is the issue with AlDente Free on your setup?

Best,
Matthias

ademlabs, 2023-11-22:
Ok so what I did with my setup previously:

 - Installed Al Dente Free
 - Set Charge Limit to 80%

The result was that it limited my charge to 80-81% all the time then stopped.
After the Sonoma update, it suddenly started to charge the computer up to 100%. I tried to uninstall and reinstall with same result.

My initial report could be erroneous as I thought Al Dente Free was stopping charge when reaching the preset limit before allowing the computer to sleep, and thought the option was activated by default.

On a default setup, should Al Dente Free ensure the charge is done until the limit is reached and not allow it to overcharge? If yes, then that's the behaviour I saw on Ventura but not after the Sonoma update.