AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1066: Sonoma 14.0 displays are off

markonikolic, 2023-10-06:
**Describe the bug**
I am using a MacBook Pro M1 with two external monitors and a closed lid. AlDente is installed and working with previous macOS versions. The power adapter is plugged in, and in battery settings, the option "Prevent automatic sleeping on power adapter when the display is off" is enabled. 

**To Reproduce**
After upgrading to Sonoma 14.0, the following happens:
- When the lid is open, the external monitor works.
- When the lid is closed the behaviour is erratic. Sometimes, monitors are disconnected after several seconds; it looks like Mac went to sleep. Sometimes the monitor starts to turn off and turn on, like going to sleep and back.
- Once the lid is open, everything works as expected.
- When AlDente is removed, everything works as expected.

**Expected behavior**
When the lid is closed, the external monitors should remain turned on.

**MacBook (please complete the following information):**
 - MacBook model (MacBook Pro, 14", 2021)
 - macOS version (Sonoma 14.0)
 - AlDente Pro
 - AlDente Version 1.22.2
 - List of all enabled/disabled settings in AlDente:
 Charge tab: "Lock screen when closed" is disabled; all others are enabled.
Features tab: Sailing mode and Heat protection are enabled

**Additional context**
Battery settings, the option "Prevent automatic sleeping on power adapter when the display is off" is enabled.

mojoba, 2023-10-08:
I have a similar issue where the external monitor doesn't go into power save mode because the usb c connection stays alive charging the macbook with AlDente running

MatthiasKerbl, 2023-10-19:
Hi @markonikolic,

Do you experience the same issues with our newest version AlDente 1.22.3?

Best,
Matthias

closed 2023-12-18 by MatthiasKerbl