AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1071: Intel Mode not working properly with Sonoma

frafer6, 2023-10-11:
Hello,
Since updating to Sonoma, if I check "intel mode" and reboot the pc, Aldente does not remember the charge threshold I set, and goes back to 100%. This bug is not present when "intel mode" is unchecked. 

 - MacBook model : MacBook Pro 2019 16'
 - macOS version: Sonoma 14
 - AlDente Free
 - AlDente Versio: 1.22.3

MatthiasKerbl, 2023-10-11:
Hi @frafer6,

Thank you for reporting this bug. We will try to fix it with the next update.

Best,
Matthias

alirzaorg, 2023-10-13:
I have the same issue tho!

kapitainsky, 2023-10-26:
I have AlDente Lauch at login enabled, in Login Items and allowed in the Background list. It used to work almost perfectly until macOS upgrade to Sonoma. Now it auto starts but battery limit is always reset to 100%.

Attach my debug log:
[aldentepro-debug-2023-10-26-12:44:46.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13176929/aldentepro-debug-2023-10-26-12.44.46.zip)

Happy to provide any additional info it it helps solving the issue.

MatthiasKerbl, 2023-10-31:
Hi everyone,

Just a short update about this issue. We were able to find the cause of it and will fix it with the next update.

Best,
Matthias

kapitainsky, 2023-11-08:
Installed v1.23 - reinstalled helper - no change. When I exit AlDente and start again limit always returns to 100%.

fraferla, 2023-11-09:
> Installed v1.23 - reinstalled helper - no change. When I exit AlDente and start again limit always returns to 100%.
> 
> Is there anything I can provide to help to fix this issue?

same here

JasSingh685, 2023-11-10:
> Installed v1.23 - reinstalled helper - no change. When I exit AlDente and start again limit always returns to 100%.

Same issue here unfortunately, even with v1.23.

MatthiasKerbl, 2023-11-10:
Hi everyone,

Thank you for reporting back and we are sorry that it still does not work correctly. We will try to fix it with the next update.

Best,
Matthias

MatthiasKerbl, 2023-12-18:
Hi everyone,

We have just released AlDente 1.24.1 which should finally fix the Intel Mode issue mentioned here.

Let us know if it works for you.

Best,
Matthias

closed 2023-12-18 by MatthiasKerbl

fraferla, 2023-12-18:
it is now working correctly for me, thank you very much!