AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1072: Sailing mode does not work on MacOS 14

baf28, 2023-10-16:
**Describe the bug**
When the macbook discharges to 78% and connects the charger, it starts charging to the 80% limit. Before the update to macOS 14, everything worked well.

**To Reproduce**
Steps to reproduce the behavior:
1. Discharge macbook by 1-9%
2. connect the charger
3. Sailing mode will not work, the battery will start charging

**Expected behavior**
Before the launch, the sailing mode worked and the macbook was not charged until the charge percentage was lower than 70%

**Screenshots**
I don't know how to upload a screenshot here, sorry

**MacBook 16 M1 Pro:**
 - macOS version 14
 - AlDente Pro
 - AlDente Version 1.22.3
 - List of all enabled/disabled settings in AlDente: Sailing mode is on

MatthiasKerbl, 2023-10-16:
Hi @baf28,

Was the Sailing Mode enabled and active while this happened? As indicated by the paper plane icon in the slider of AlDente?

[image]

Also, the Sailing Mode usually starts from the upper limit, therefore, it will always charge to the upper limit first to trigger the Sailing Mode.

Best,
Matthias

baf28, 2023-10-16:
No, such an icon was not visible, but the mode was enabled in the settings, I checked it for sure.
[image]
Everything used to work as expected, now not unfortunately (((

closed 2023-10-16 by MatthiasKerbl

baf28, 2023-10-16:
You're right, thank you. I was very worried about this moment and...
I turned off the charging, waited for 78% and turned it on. Sailing mode works!!!
[image]
Sorry to bother you, next time I'll be more restrained, I was just scared..