AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1074: AIdente is not working properly.

xinglinquan, 2023-10-19:
**Describe the bug**
I have set the upper limit of charging power to 80%. When using the software, everything usually meets expectations. However, one night after that, I realized that the power had increased to 88%. When I opened the system settings, I found that the power suddenly surged from 80% to 88% during a certain period of time. From the graph, it can be seen that there was a sudden increase without a slope. I have also noticed a situation where it fell to 79% before, but I think this is normal, but the amplitude is too large this time, It's still a mutation, I feel something's wrong and I hope to get an answer.

**Screenshots**
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook model: macbook pro 14 m2pro(2023) 
 - macOS version :macOS 13.5.1
 - AlDente Pro or AlDente Free:AlDente Free
 - AlDente Version:1.22.3
 - List of all enabled/disabled settings in AlDente:
[image]
[image]
[image]

MatthiasKerbl, 2023-10-19:
Hi @xinglinquan,

jumps like that in the state of charge/battery percentage are usually a sign of a battery miscalibration. Therefore, I would recommend running the Calibration Mode of AlDente Pro or doing a manual calibration by cycling the battery between 0-100% twice.

Best,
Matthias

MatthiasKerbl, 2023-10-20:
> Thank you！ So is it also due to battery calibration that the software did not limit the charging limit according to the settings? (80%->88%) Also, if it's really a battery calibration issue, is it a bit strange and surprising? I only bought my computer in less than two months.

Hi @xinglinquan,
It is difficult to say what exactly the reason is that your state of charge jumped from 80%-88%. It looks like it didn't charge but rather just "jumped" up to 88% which from our experience can be caused by miscalibration.
This behavior is definitely strange and a miscalibration after only two months is definitely uncommon.

Best,
Matthias

xinglinquan, 2023-10-22:
Thank you very much, but I still have some concerns.
1. Just now there was the situation I mentioned earlier, where the battery level dropped from 87% to 86%. Considering the previous situation where the battery level dropped from 80% to 79%, what is the reason for this problem, and is it also related to battery calibration? Both scenarios are running AIdente, and the maximum charging limit is set to 80%.
2. Is it possible that all the problems I have encountered before ("jumped" up to 88% or decrease of 1% in battery power) were caused by AIdente software, or indirectly by AIdente (i.e. AIdent causing battery calibration issues)?
3. I would also like to know if these issues will (or have) caused adverse effects or damage to my computer battery?

xinglinquan, 2023-10-26:
Hi, just now, I noticed that the same situation happened again, with the battery jumping from 80% to 89%, which is really frustrating.

xinglinquan, 2023-11-02:
Thank you very much for your patient answer!

closed 2023-11-10 by MatthiasKerbl