AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1075: Show power drawn by USB devices in power-flow visualisation

Jofkos, 2023-10-20:
**Is your feature request related to a problem? Please describe.**
When looking at the power flow (in the pop-up), it only differentiates between power flowing to/from the battery and power to/from the device. When plugging in an external device like, e.g., a phone, this all gets lumped under the power drawn by the system.

**Describe the solution you'd like**
An additional category of where the power flows to, representing external devices drawing power via USB/Thunderbolt (e.g., external hard drive or phone).

**Describe alternatives you've considered**
/

**Additional context**
/

MatthiasKerbl, 2023-10-20:
Hi @Jofkos,

Great idea, we already have a feature request open for this here #1002.

Best,
Matthias

closed 2023-10-20 by MatthiasKerbl

Jofkos, 2023-10-23:
Apologies, I hadn't seen that there already was an open issue. Glad to hear someone else had the same idea!