AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1076: when i reboot my mac, aldante's setting is reset.

EMJ97, 2023-10-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-10-24:
Hi @EMJ97,

Do you have Launch at login enabled?
Do you have AlDente in the Login Items in the macOS settings?
Do you have AlDente enabled in the Login Items setting in macOS in the Allow in the Background list?

Please share a debug file from when you restart the MacBook but the settings have reset.

Best,
Matthias

kapitainsky, 2023-10-26:
I have AlDente Lauch at login enabled, in Login Items and allowed in the Background list. It used to work almost perfectly until macOS upgrade to Sonoma. Now it auto starts but battery limit is always reset to 100%.

Attach my debug log:

[aldentepro-debug-2023-10-26-12:44:46.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13176915/aldentepro-debug-2023-10-26-12.44.46.zip)

Happy to provide any additional info it it helps solving the issue.

alex-lechner, 2023-10-27:
I have the exact same problem! 

Also, @MatthiasKerbl I can answer all three questions with yes. I even reinstalled the app but nothing worked.
The issue came after I updated to Sonoma.

I'm on a Macbook Pro, 2,4 GHz 8-Core Intel Core i9 with 32 GB RAM, and Sonoma 14.0

MatthiasKerbl, 2023-10-31:
Hi @alex-lechner and @EMJ97,

Thanks for sharing this issue and we are sorry you are experiencing this. We are still investigating why this is happening to some users. Also, there is already an issue open about this problem here #898. We will keep you updated about this issue there.

Best,
Matthias

closed 2023-10-31 by MatthiasKerbl