AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1078: Power Flow is not available on this system

Drammy, 2023-10-27:
Power flow reports itself as unavailable since upgrading to Sonoma

**To Reproduce**
1. Go to 'Settings'
2. Click on 'Stats'
3. Observe notice where the power flow chart should be... `Due to technical restrictions, Power Flow is not available on this system.`

**Expected behavior**
I expect to see the power flow graphic

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 15" 2019, 2.4 GHz 8-Core Intel Core i9, 32 GB 2400 MHz DDR4, Radeon Pro Vega 20 4 GB, Intel UHD Graphics 630 1536 MB
 - macOS - 14.1 (23B74)
 - AlDente Pro 1.22.3
 - List of all enabled/disabled settings in AlDente:
[image]
[image]

MatthiasKerbl, 2023-10-31:
Hi @Drammy,

Thank you for this bug report and we are sorry you are experiencing this issue. In order to assess this problem properly, could you send us a debug file? _[How to generate and share a debug file](urlhttps://apphousekitchen.com/how-to-generate-and-share-a-debug-file/)_

Best,
Matthias

Drammy, 2023-10-31:
Thanks @MatthiasKerbl
I've generated the debug file but can't find an email with the details of the support email address..?

Drammy, 2023-10-31:
Yeah I saw that on the knowledge-base article you shared.  Unfortunately, I think I must have been a bit heavy-handed during an inbox clear-out.

I've attached the file here:
[aldentepro-debug-2023-10-31-09:59:22.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13215159/aldentepro-debug-2023-10-31-09.59.22.zip)

Drammy, 2023-10-31:
Have found the email (outlook search is rubbish).  Thanks

MatthiasKerbl, 2023-11-01:
Hi @Drammy,

Good news, we were able to find the cause of this bug and will fix it with the next update. 

Best,
Matthias

Drammy, 2023-11-01:
That's great, thanks @MatthiasKerbl

MatthiasKerbl, 2023-11-09:
Hi everyone,

We have just released AlDente 1.23 which includes a fix for this issue. To get AlDente 1.23 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Cheers,
Matthias

closed 2023-11-09 by MatthiasKerbl

Drammy, 2023-11-14:
Power flow is now working, thanks