AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1079: Travel mode

EdogawaKun, 2023-10-29:
I was wishing if it's possible to make "travel mode", where the MacBook will keep charging up to 100% until when the user turns off the switch.

I know just quitting AlDente and not opening it will achieve the same thing, but an in-app option, I think, will be a more elegant solution.

Thanks for the great app!

DamonHD, 2023-10-29:
I do exactly this by activating Top Up (+).

I also use this for filling up from my off-grid system when the sun is out.

When I unplug from the wall or the sun goes behind a big cloud, top-up mode stops and my normal 80% limit cuts in.

MatthiasKerbl, 2023-10-31:
> I was wishing if it's possible to make "travel mode", where the MacBook will keep charging up to 100% until when the user turns off the switch.
> 
> I know just quitting AlDente and not opening it will achieve the same thing, but an in-app option, I think, will be a more elegant solution. Also, I love the power flow animations so much.
> 
> Thanks for the great app!

Hi @EdogawaKun,

As mentioned by @DamonHD, this feature is already available in AlDente and is called Top Up. If you are traveling for a longer period and want to charge to 100% more often, just change the charge limit to 100% for this period.

Best,
Matthias

closed 2023-10-31 by MatthiasKerbl