AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1083: Sailing mode and calibration stopped working completely after Sonoma update

ThrobbyQuelch, 2023-10-31:
**Describe the bug**
Sailing mode fails to let battery drop percentage under the limit. It only drops down from say, 100% to my desired limit but doesn't sail any lower, despite the setting being on.

Calibration has the same problem, not being able to let the battery discharge below set limit.

**To Reproduce**
Update to Sonoma I guess

**Expected behavior**
Sailing mode to let the battery discharge from 80% to 60%

**MacBook (please complete the following information):**
 - MacBook model: M1 Pro
 - macOS version Sonoma 14.0
 - AlDente Pro 
 - AlDente Version 1.22.3
 - List of all enabled/disabled settings in AlDente: Auto updates ON, Automatic Discharge ON, Disable Sleep until Charge Limit ON, Sailing Mode ON, Heat Protection ON

**Additional context**
I reinstalled the helper, rebooted, updated all there was to update.

MatthiasKerbl, 2023-11-10:
Hi @ThrobbyQuelch,

This seems to be a known bug documented in this issue #768.

In short, this is how you should be able to fix it:

AlDente stopped working after the macOS update!

Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.

To get AlDente to work again, please try the following steps which have worked for many users before:

Unplug and discharge your MacBook to less than your current charge limit in AlDente
Turn off your MacBook and close the lid for a couple of minutes
Unplug the power brick from the outlet and wait a couple of minutes
Turn on your MacBook and plug it in again
Check if it stops charging at the charge limit set in AlDente
We have been aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Best,
Matthias

ThrobbyQuelch, 2023-11-10:
Hi, yes, I tried this, didn't really help. The limiter stops at 80% but it will not sail down no matter what, despite the sailing icon and charge pause icon being shown.

MatthiasKerbl, 2023-11-10:
This is actually the intended behavior of the Sailing Mode.
It is a common misconception that the Sailing Mode does actively charge and discharge your MacBook between the sailing range. This would actually be much worse for the battery.
You can read a detailed description of what the Sailing Mode actually does on our blog "[Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)"

Best,
Matthias

ThrobbyQuelch, 2023-11-11:
Yes, I'm aware, but shouldn't it let the battery discharge normally with use until the desired %? That was the way it worked before, now it just keeps it charged to the limit no matter how heavy the usage. As soon as I physically unplug it, it discharges normally.

MatthiasKerbl, 2023-11-11:
> Yes, I'm aware, but shouldn't it let the battery discharge normally with use until the desired %? That was the way it worked before, now it just keeps it charged to the limit no matter how heavy the usage. As soon as I physically unplug it, it discharges normally.

If you mean by discharge normally that while you use the MacBook, the energy is coming from the battery, that's not the case. During Sailing Mode, the energy is coming from the plug. The battery is only losing charge due to self-discharge over time buffering in some cases. Therefore, the battery percentage does barely go down, depending on the certain scenario.

Best,
Matthias

ThrobbyQuelch, 2023-11-12:
Oh okay, guess I never noticed before. Thank you!

closed 2023-11-12 by MatthiasKerbl