AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1086: Time remaining till battery depletion

FelixKLG, 2023-11-06:
**Is your feature request related to a problem? Please describe.**
I've had 0 issues with AlDente since installing it, it has worked perfectly. I would however like a way to view the estimated time remaining from my charge from the menu bar pop-out like I can in iStat Util's battery monitor or via Activity Monitor.

**Describe the solution you'd like**
I'd like the estimated time remaining until my battery dies visible from the AlDente pop-out menu.

**Describe alternatives you've considered**
iStat or checking it from Activity Monitor.

**Additional context**
[image]
[image]
[image]

DamonHD, 2023-11-06:
Seconded.

Then I could stop using one of the other battery utilities I use!

Rgds

Damon

TolleyLikesRice, 2023-11-07:
More detailed duplicate of #1050 ?

Would be great though, then I wouldn't need to have the aldente icon hidden my bartender.

MatthiasKerbl, 2023-11-09:
Hi @FelixKLG,

Good idea, this is definitely something we will look into in one of the future updates. Also, since #1050 already mentions this, I will close this issue.

Best,
Matthias

closed 2023-11-09 by MatthiasKerbl

jrhorne, 2024-01-15:
@FelixKLG related question— has the iStat time remaining estimate been accurate while using AlDente? Was going to stick it into my menu bar but curious if it's been somewhat reliable