AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1087: Charge limit does not work at all on MacBookPro11,3 on macOS 12.7.1

Piipperi, 2023-11-06:
**Describe the bug**
AlDente just doesn't do anything on this computer. I am not sure if my config is supported as the supported Mac and macOS versions are not listed anywhere on AlDente's site. The charge limit doesn't work (the battery charges above the limit at doesn't stop) and the Discharge button doesn't do anything neither.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a charge limit
2. Discharge and recharge the laptop
3. See how it hasn't stopped at the charge limit

**Expected behavior**
I'd expect it to not keep charging past a limit I've set in AlDente.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model MacBook Pro (Retina, 15-inch, Late 2013)
 - macOS version 12.7.1
 - AlDente Free
 - AlDente Version 1.22.3
 - List of all enabled/disabled settings in AlDente: 80% Limit is enabled. No further settings have been changed.

**Additional context**
I am using OpenCore Legacy Patcher on this laptop, but since AlDente doesn't prompt me about any errors, it's not easy to say if this is the cause. I have already tried PRAM resets and reinstalling AlDente.

theking988, 2023-11-08:
macbook 12 2017. also not working by 1.22.3.   The old version is no problem.

MatthiasKerbl, 2023-11-10:
Hi @Piipperi,

Could you share a debug file for us to assess your issue better?
You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

Piipperi, 2023-11-21:
[aldentepro-debug-2023-11-21-13:37:17.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13425923/aldentepro-debug-2023-11-21-13.37.17.zip)

MatthiasKerbl, 2023-11-22:
> [aldentepro-debug-2023-11-21-13:37:17.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13425923/aldentepro-debug-2023-11-21-13.37.17.zip)

Hi @Piipperi,

Thank you for sharing the debug file. Unfortunately, we are not sure what exactly the issue is in your case. Therefore, we are recommending to perform an SMC reset. Just follow this guide by Apple ["Reset the SMC of your Mac"](https://support.apple.com/en-us/HT201295).

Best,
Matthias

Piipperi, 2023-12-14:
I tried doing SMC + PRAM reset, and now discharge function works, but the charging still doesn't seem to stop at 80% (where I have set the limit at)

MatthiasKerbl, 2023-12-16:
> I tried doing SMC + PRAM reset, and now discharge function works, but the charging still doesn't seem to stop at 80% (where I have set the limit at)

Hi @Piipperi,

What are the exact steps you are taking when the charge limit does not work?
Also, can you share a debug file again directly after the charge limit not working?

Best,
Matthias

Piipperi, 2023-12-27:
I did more testing and it appears that my battery's BMS is just broken. In situations like these though I do wish the Auto-Discharge feature wasn't a paid one.

closed 2023-12-27 by Piipperi