AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1088: Sleep not prevented when lid closed after updating to Sonoma 14.1

TolleyLikesRice, 2023-11-07:
**Describe the bug**
After starting charging with the lid open, charging then stops when the lid is closed. Only had this issue since updating to Sonoma 14.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Open lid
2. Plug in & remember percentage
3. Close Lid
4. Wait for a while
5. Open lid and check charge level.

**MacBook (please complete the following information):**
 - 2023 14" M2 MBP
 - Sonoma 14.1
 - AlDente Pro
 - AlDente Version: 1.22.3
 - Settings in AlDente:
[image]
[image]

TolleyLikesRice, 2023-11-07:
Also:
[image]

TolleyLikesRice, 2023-11-07:
Just done a little bit more testing, so if I leave the lid open, it seems to stay awake just fine, it's just not staying awake once the lid has been closed.

MatthiasKerbl, 2023-11-10:
Hi @TolleyLikesRice,

I am currently not able to reproduce this issue.
Are you also experiencing this issue on our newest version AlDente 1.23?
Also, is the sleep option greyed out in macOS before you close the lid?

Best,
Matthias

TolleyLikesRice, 2023-11-10:
Seems to be fixed in 1.23! Thank you!

TolleyLikesRice, 2023-12-06:
Hang, no it's a different issue, sorry.