AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1091: Charge limiter not working after macOS 13.6.1 on MacBook 12 (2017)

robertoarlt, 2023-11-08:
**Describe the bug**
Despite reinstalling the helper app after system upgrade and changing the Aldente settings the battery charges to full capacities

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBook (Retina, 12-inch, 2017)  (https://support.apple.com/kb/SP757?viewlocale=en_US&locale=en_US)
 - macOS version 13.6.1
 - AlDente Pro 
 - AlDente Version 1.22.3
 - List of all enabled/disabled settings in AlDente:
[image]

**Additional context**
Add any other context about the problem here.

hw0603, 2023-11-08:
I'm experiencing same situation on my Macbook 12 (2017), Aldente Free.
When I manully turn on the discharing mode, it also doesn't work

MatthiasKerbl, 2023-11-10:
Hi @robertoarlt,

This seems to be a known bug documented in this issue #768.

In short, this is how you should be able to fix it:

AlDente stops working after the macOS update!

Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.

To get AlDente to work again, please try the following steps which have worked for many users before:

1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
2. Turn off your MacBook and close the lid for a couple of minutes
3. Unplug the power brick from the outlet and wait a couple of minutes
4. Turn on your MacBook and plug it in again
5. Check if it stops charging at the charge limit set in AlDente

We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Best,
Matthias

MatthiasKerbl, 2023-12-19:
Hi @robertoarlt,

Are you still experiencing this issue?

Best,
Matthias

robertoarlt, 2023-12-19:
Hello Matthias

After the recent update all seems to work fine again. Many thanks for your
help and support and efforts.

All the best

Martin

closed 2023-12-19 by MatthiasKerbl