AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1097: AlDente pro purchase problem

robertdulca204, 2023-11-10:
**Describe the bug**
So I bought AlDente pro lifetime yesterday ( 9 th of November 2023) and after not receiving any email, I emailed the support section. An automated message was received in which said that I should go talk to a bot to resolve my issue. The issue is that I typed my email wrong from the start (I miss typed email@yahoo.com instead of email@icloud.com (correct one)). The bot said that he forwarded my message to someone, but I still didn't got any reply. I have all the screenshots with the payment and everything.

I expected an easier problem solving with this matter.

I will attach screenshots with the screen after you complete the payment. 
[image]
[image]
(I won't attach right now the payment sheet from my bank, for security reasons.)

**MacBook (Pro 2021):**
 - macOS version (Sonoma 14.1.1)
 - AlDente Pro lifetime

MatthiasKerbl, 2023-11-10:
Hi @robertdulca204,

First of all, thank you for buying AlDente Pro and I am sorry to hear that the support of our merchant of record Paddle is taking longer than expected. However, they should take care of this issue within a couple of days.

Best,
Matthias

closed 2023-11-10 by MatthiasKerbl

robertdulca204, 2023-11-10:
Hi Matthias,

Pleased to se that someone answered to my problem, I hope they will take care of this issue, I explained everything in the email I sent to them is just an email, I sent all the required documents to prove that I did the purchase ( card identification and even screenshots of the bank statement and everything).

Best
Robert Dulca