AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1099: M1 Macbook Air [Unsual Heating]

hemanG-G, 2023-11-14:
I'm observing unusual heating specifically around the charging port area , while its plugged in and charge limit at 75%, is this normal ?  
Why is it heating specifically around that area , Is there chance of damage to charging circuitry .

MatthiasKerbl, 2023-11-15:
Hi @hemanG-G,

In general, some degree of heating around this area is normal. However, it should not be too hot. Do you only experience this area heating up while using AlDente?

Best,
Matthias

hemanG-G, 2023-11-16:
Its not too hot , but yeah  its mostly around the charging port area  ,if thats not the CPU location it might be concerning.

Does Aldente use the same method that Apple uses when we charge it to 100 and leave it plugged in to maintain the charge , or is  it a third party method of maintaining charge the charge.

I'll test it for a few more days , to be sure .

hemanG-G, 2023-11-21:
The heating was present to some extent when i kept it plugged in at 100  without AlDente , but i'm still skeptical of the specific heating around the charging port area .

Closing the Issue , Thanks @MatthiasKerbl

closed 2023-11-21 by hemanG-G