AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1102: Option activation not possible

sando31877, 2023-11-17:
[image]
[image]

decorating options can be activated, but functional options can't be activated.
like AlDente Free

Even after reinstall AlDente, the existing option is applied.
I think it's because it hasn't been fully reinstalled.

How can I do a complete reinstallation?
or how can I fix a problem where option activation is not possible?

AlDente Pro 1.23
2019 macbook pro 16"
macOs Sonoma 14.1.1

MatthiasKerbl, 2023-11-17:
Hi @sando31877,

I am not sure if I understand your problem completely.
Do you mean by "option activation is not possible" that most of the features are greyed out? This is because you have the Intel Model activated. You can read more about the Intel Mode on our blog here: ["Feature Explanation: Intel Mode"](https://apphousekitchen.com/feature-explanation-intel-mode/)
If you are not using Windows via Bootcamp, you can deactivate the Intel Mode and use the rest of the features of AlDente Pro normally.

Best,
Matthias

sando31877, 2023-11-17:
> Hi @sando31877,
>
> I am not sure if I understand your problem completely. Do you mean by "option activation is not possible" that most of the features are greyed out? This is because you have the [...]

I didn't realize it was grayed out when I turned on Intel mode.
Thanks to you, I can now use aldente properly.

closed 2023-11-17 by MatthiasKerbl