AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1107: Charge limit set to 100% after calibration finished

mike-ward, 2023-11-21:
**Describe the bug**
Charge limit set a 70%. Ran calibration. After calibration finished, charge limit remained at 100%. Expected charge limit to return to 70%.
  
  Model Name:	MacBook Air
  Model Identifier:	Mac14,15
  Model Number:	MQKW3LL/A
  Chip:	Apple M2
  Total Number of Cores:	8 (4 performance and 4 efficiency)
  Memory:	8 GB
  System Firmware Version:	10151.41.12
  OS Loader Version:	10151.41.12

AlDente Pro 1.23

[image]

[image]

MatthiasKerbl, 2023-11-22:
Hi @mike-ward,

Unfortunately, we are currently not able to reproduce this issue. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

mike-ward, 2023-11-22:
[aldentepro-debug-2023-11-22-07:24:04.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13440002/aldentepro-debug-2023-11-22-07.24.04.pasta.zip)

Pasta files? Too funny.

MatthiasKerbl, 2023-11-23:
Hi @mike-ward,

In the debug file it seems like the Calibration Mode was not activated. Could you generate and share a debug file again directly after this bug happens for you?

Thanks,
Matthias

mike-ward, 2023-11-28:
Ran the calibration with 1.24 Pro. Worked as expected this time and returned to the 70% charge limit after completion.

Here's the debug file in case you're interested in reviewing the calibration run.

[aldentepro-debug-2023-11-28-08:30:06.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13489286/aldentepro-debug-2023-11-28-08.30.06.pasta.zip)

closed 2023-11-28 by mike-ward